br3ttb/Arduino-PID-Library #31: Calculation error

rtlopez, 2015-10-20:
Hi,

Thanks for the cool and useful library. 
I'm no expert but it seems to me that the Compute method has a bug. Specifically, it concerns line
`double output = kp * error + ITerm- kd * dInput;`
it seems to me that it should look like this
`double output = kp * error + iTerm + kd * dInput;`
for me it results directly from the formula
https://en.wikipedia.org/wiki/PID_controller
I would add that I am not entirely sure, and I have not tested it. Maybe the tests show that current version works better. If I'm wrong or missing something, please reject this issue.
Generally good job ;)

isplitt, 2015-10-20:
yes, the variables are maybe not named correctly, but the algo is OK. Every now and then someone stumples upon this. have a look at #28 (and maybe #28).

br3ttb, 2015-10-20:
Please see here: http://brettbeauregard.com/blog/2011/04/improving-the-beginner%e2%80%99s-pid-derivative-kick/

br3ttb, 2015-10-20:
@split it's named dInput, not dEer. How would you suggest the variable be named?

br3ttb, 2015-10-20:
@split=>@isplitt (on my phone)

closed 2015-10-20 by br3ttb

isplitt, 2015-10-20:
maybe just call it "dNOTErr" :-) 
I think it's ok how it is, but it still confuses some people from time to time. 
Another solution could be to spend this one OP and tranform it to dErr.

rtlopez, 2015-10-20:
Hi,

Thanks for quick reply. You are right, dError and dInput aren't same. As I understood, this optimisation cause that the derivative component responds only to a distortion of the controlled object but does not responds when the setpoint change. Sometimes it could be useful, I have to test it in my project. It is good to know how the controller works. Have you thought about add an option to customize this behaviour?

Cheers.

br3ttb, 2015-10-20:
It would be very easy to customize it's true, but I have yet to find a
situation where the kick is beneficial.  This isn't to say there isn't one,
just that I have never seen it. (And I've seen a LOT of loops)