br3ttb/Arduino-PID-Library #35: ITerm cannot be removed when tuning.

JacobChrist, 2015-12-30:
Since 

```c++
ITerm += ki * error 
```

and 

```c++
output = kp * error + ITerm - kd * dInput;
```

The i term can never be shut off.

Although Ki can be thought of as a constant of integration (thus it doesn't matter if it is inside or out side of the integral) it actually is not constant and can be adjusted by the user.  If your tuning a PID loop and want to remove the i term by setting i to 0 your out of luck since the ITerm is never removed from the output equation.  Here is a suggested fix:
```c++
      /*Compute all the working error variables*/
      double input = *myInput;
      double error = *mySetpoint - input;
      ITerm += error;
      double ki_x_ITerm = ki * ITerm
      if(ki_x_ITerm > outMax) ITerm= outMax;
      else if(ki_x_ITerm < outMin) ITerm= outMin;
      double dInput = (input - lastInput);
 
      /*Compute PID Output*/
      double output = (kp * error) + (ki_x_ITerm) - (kd * dInput);
```

This should be functionally equivalent to existing code with the exception when i is set to 0.  I'm testing this now.

Jacob

JacobChrist, 2015-12-31:
I read this:

http://brettbeauregard.com/blog/2011/04/improving-the-beginner%E2%80%99s-pid-tuning-changes/

So if this feature is desired, then an if would be needed remove the summed error when i = 0. Hmmm.

br3ttb, 2015-12-31:
the ITerm variable does double duty as both the Integral term and the baseline (bias) value.  when the controller goes from manual=>auto, it is initialized to whatever the Output value is.  From there, it changes based on the value of Ki.  if Ki is 0, it remains constant.

so, if you want ITerm=0 (your pid to be working from an unchanging baseline of 0,)  make ki=0, and set the output variable to 0 before putting your pid in manual.

closed 2015-12-31 by br3ttb

JacobChrist, 2015-12-31:
Okay, I see what you did here.   It looks like if ki != 0 and you are already in AUTOMATIC mode then you need to go to MANUAL set ki to zero then back to AUTOMATIC to reset the ITerm.  I'm okay with this.  Its a little more effort on my side, but I think I really like the ability to change ki on the fly without the bump (even though its not affecting me).

br3ttb, 2015-12-31:
If you change ki on the fly there will be no bump. It's only if you want
the baseline output to be zero that you need to do something special.

The special thing you need to do is make output=0 while the pid is in
manual.

Tunings can be changed in the fly without a problem at any time.