br3ttb/Arduino-PID-Library #37: iMax

svefro, 2016-01-06:
Hi. i have just swapped over from my own pdi implementation to yours. It's working fine and is a lot simpler code wise than what i had mashed together. However i'm missing an iMax parameter. i'm now using it for temperature control on a slow system that has to change temperatures at certain times. (Beer brewing). for this i need a relative small i value to keep the temp constant. but when i increase the setpoint the i term starts to build and when the setpoint is reached it takes to long to settle down again. for this i have implemented a i max parameter. Is this something you would like to have also or should i just keep it in my private fork?

I have added :
double GetIterm()
double GetIMax()
setIMax(double)
double getIMax()

in compute() ITerm validates to IMax and not outMax.
when outMax or IMax is changed i check to see that IMax <= outMax

i would have sent you a compare link but i used a different encoding and all the lines are green.

i'm unsure how this works on a reversed controller since i don't have anything to test with.

If you would like this i can make a proper pull request.

one more thing. i have made SetIMax to set double IMax but i think this should be set in either SetOutputLimits() or SetTunings() what do you think?

br3ttb, 2016-01-07:
If Imax it's working for you that's great, but I feel that the problem it
solves (pid overshoot on an integrating process) can be addressed in other
ways that do not increase the complexity of the library code.

You could, for example, put the pid in manual at 100% output when you are
far below setpoint, then drop to 0% and switch to auto when you are close.

Ultimately the most elegant solution is to use something known as
"proportional on measurement," but that will have to wait until I write an
advanced pid library.

svefro, 2016-01-07:
I have seen IMax on many PID controllers even dMax. on some. I'm happy to have this only in my fork. but it don't add much complexity to the code in my opinion.
On my temp controller i set d max to a bit above the maximum heat loss i can expect and use a relatively small i value. P50 i=0.02 D = 300
Attached is an image after implementing IMax.
Before i had large overshots or difficulty keeping the correct temp. (or a way to slow temp increese.)
Now i can keep the temperature to within Max 0.5 deg mostly 0.2deg. Celcius
[image]

by the way. Great Library! :)

br3ttb, 2016-01-08:
That looks great!  As far as Imax goes, it's not something you find in
industrial pids. My started goal with this library was to create something
on par with those controllers.

Another option that I didn't mention in my last post would be to call
setoutputlimits at runtime. If you clamped your max output when far below
setpoint I think you would be able to achieve nearly identical results.

Ultimately however, this is the benefit of open source. You have made a
solution that works for you!