br3ttb/Arduino-PID-Library #38: Bumpless Mode Switching is not ensured

murmlgrmpf, 2016-02-04:
Hi,
Thank you for the excellent pid library! I have noticed that the way the iterm is initialized when switching from manual to automatic mode does not ensure a bumpless transfer because the kp*error term is not considered by the initialization.
Is it a bug? Is it a feature? It makes my life much easier, as I am using ki=0 and use the initialization to generate a static offset...

Thomas

br3ttb, 2016-02-04:
That's the expected behavior.  the output jump you're seeing is due to the
fact that you have some error when you initialize the controller (setpoint
not equal to Input.)  so the baseline (ITerm) is initialized to the current
output, but you then have some proportional error that gets added to it.

if I were to make this situation bumpless, (on initialization ITerm =
output - Kp*err or whatever) that would lead to a more complicated user
problem when Ki=0, as the iterm wouldn't adjust, and you've have a fixed
offset.

SO.  with this library (as with many industrial models,)  if you initialize
the controller and Input = Setpoint there will be no bump.  if you are not
at setpoint, you will see a bump that is not due to internal weirdness, but
just the pid trying to get you to setpoint as it should.

murmlgrmpf, 2016-02-04:
Thank you for the clarification. I had already half way expected that it is a feature.

closed 2016-02-04 by murmlgrmpf