br3ttb/Arduino-PID-Library #40: The variable Output always remains 0

sriranjanr, 2016-03-07:
Hello,

I downloaded the library and tried to run the examples.I am printing the variables Input and Output on the serial monitor. The "Output" variable always reads 0. Could you tell me why the pidcompute is not able to generate the Output?

orgua, 2016-03-07:
Could you paste your code here? When it does not work it is unlikely a problem with the lib

sriranjanr, 2016-03-07:
`#include <PID_v1.h>

#define PIN_INPUT 0
#define PIN_OUTPUT 3

//Define Variables we'll be connecting to
double Setpoint, Input, Output;

//Define the aggressive and conservative Tuning Parameters
double aggKp=4, aggKi=0.2, aggKd=1;
double consKp=1, consKi=0.05, consKd=0.25;

PID myPID(&Input, &Output, &Setpoint, consKp, consKi, consKd, DIRECT);

void setup()
{
  Serial.begin(115200);
  //Specify the links and initial tuning parameters
   pinMode(3,OUTPUT);
  //initialize the variables we're linked to
  Input = analogRead(PIN_INPUT);
  Setpoint = 100;
  
  //turn the PID on
  myPID.SetMode(AUTOMATIC);
}

void loop()
{
  Input = analogRead(PIN_INPUT);

  double gap = abs(Setpoint-Input); //distance away from setpoint
  if (gap < 10)
  {  //we're close to setpoint, use conservative tuning parameters
    myPID.SetTunings(consKp, consKi, consKd);
  }
  else
  {
     //we're far from setpoint, use aggressive tuning parameters
     myPID.SetTunings(aggKp, aggKi, aggKd);
  }

  myPID.Compute();
 // analogWrite(3, (int)(Output));
  analogWrite(3,0);
  Serial.print("Input ");
  Serial.print(Input);
  Serial.print(" Output ");
  Serial.println((Output));
}

`

orgua, 2016-03-07:
code is not pretty, but ok. can you vary the input from 0-1024 and confirm that nothing happens?

captamazing, 2016-10-20:
I had this problem also when using a negative derivative term. Seems like that is not the cause for you, but the root of the problem may be the same. I'll investigate further if I get a chance.

Scalpel78, 2016-10-29:
I have the same issue as well. The Input varies, but Output is always the minimum value.

JacobChrist, 2016-10-29:
Set Ki and Kd to zero and see if anything works.

Your analogWrite is writing 0, but I suspect that you know this and mean that when you print the Output it is zero too.

closed 2017-02-15 by br3ttb

storca, 2020-08-24:
Hello and sorry for excavating this issue 4 years later

The issue for me was the same, the output was always 0 regardless of the input and the tunings.

The solution for me was to comment the line 60 in PID_V1.cpp eg

```cpp
//if(!inAuto) return false;
```
For some reason this variable was set to the incorrect value and each time the Compute function was called, it would just return 0 and compute nothing.

br3ttb, 2020-08-24:
If inauto is false, the pid is in manual mode. Try calling
setmode(automatic)

storca, 2020-08-24:
Oops, I forgot that line... Thanks for the help!