br3ttb/Arduino-PID-Library #45: Only one PID instance supported?

Pecacheu, 2016-06-20:
I'm trying to use the library to control a Quadcopter, and it looks like only the first PID instance declared actually works. For example, if I declare them in this order, only the yaw controller works:
```
PID yawPID(&yaw, &setYaw, &rxYaw, YAW_KP, YAW_KI, YAW_KD, REVERSE);
PID pitchPID(&pitch, &setPitch, &rxPitch, PITCHROLL_KP, PITCHROLL_KI, PITCHROLL_KD, REVERSE);
PID rollPID(&roll, &setRoll, &rxRoll, PITCHROLL_KP, PITCHROLL_KI, PITCHROLL_KD, REVERSE);

//Initialize PID Controller:
void initPID() {
  yawPID.SetOutputLimits(YAW_MIN, YAW_MAX); yawPID.SetSampleTime(PID_CYCLETIME);
  pitchPID.SetOutputLimits(PITCHROLL_MIN, PITCHROLL_MAX); pitchPID.SetSampleTime(PID_CYCLETIME);
  rollPID.SetOutputLimits(PITCHROLL_MIN, PITCHROLL_MAX); rollPID.SetSampleTime(PID_CYCLETIME);
  yawPID.SetMode(AUTOMATIC); pitchPID.SetMode(AUTOMATIC); rollPID.SetMode(AUTOMATIC);
}
```
Putting pitch or roll first makes them work instead. Here is where I'm calling the Compute functions:
```
if(yawPID.Compute() || pitchPID.Compute() || rollPID.Compute()) {
  DBG.print("PID: YAW="); DBG.print(setYaw);
  DBG.print(", PITCH="); DBG.print(setPitch);
  DBG.print(", ROLL="); DBG.println(setRoll);
}
```
The board I'm using is an Arduino Zero. Also, I agree with #39 since the Zero and Due treat doubles as 64-bit double-percision, and unfortunately neither have a floating-point unit.

Pecacheu, 2016-06-20:
After messing around with the variables, I think I found the problem. Where there are multiple PID controllers, the SampleTime has to be more than 4 times slower than the loop is running. I had a `delay(50)` in there so I could read the output from the Arduino. So my `PID_CYCLETIME` constant has to be 200 or more. The strange thing is that this isn't a problem when using only one PID controller. Weird.

hildred, 2016-06-20:
or lazy evaluates. try this:

```
if(yawPID.Compute() + pitchPID.Compute() + rollPID.Compute()) {
  DBG.print("PID: YAW="); DBG.print(setYaw);
  DBG.print(", PITCH="); DBG.print(setPitch);
  DBG.print(", ROLL="); DBG.println(setRoll);
}
```

Pecacheu, 2016-06-20:
Oh! Thanks, that explains it. I forgot that OR doesn't bother computing the second value if the first one was true. (I knew that code was too simple to be true.) This works properly:
```
bool newY = yawPID.Compute(), newP = pitchPID.Compute(), newR = rollPID.Compute();
if(newY || newP || newR) {
  #ifdef DEBUG_PID
  DBG.print("PID: YAW="); DBG.print(setYaw);
  DBG.print(", PITCH="); DBG.print(setPitch);
  DBG.print(", ROLL="); DBG.println(setRoll);
  #endif
}
```

closed 2016-06-20 by Pecacheu

xoxota99, 2018-03-26:
OMG. Can we promote this fix to be somewhere in the README? This is solving a longstanding problem for me...