br3ttb/Arduino-PID-Library #46: Bug? PID Output is wrong?

nanahuse, 2016-06-22:
In this library, PID Output is calculated from
`output = kp * error + ITerm - kd * dInput`
(At line 63 in PID_v1.cpp)

But, I think  PID Output is calculated from the following formula using previous step's output  in discrete implementation.
`output =  old_output + kp * error + ITerm - kd * dInput`

reference: https://en.wikipedia.org/wiki/PID_controller#Discrete_implementation

orgua, 2016-06-22:
look at #43 #42 #36 #28

nanahuse, 2016-06-22:
Thank you for your prompt reply.
I have looked those issues. But, my issue don't resolve.

On example  PID_Basic, the output calculated by  PID class is used as analogWrite value.
And on example PID_RelayOutput, the output is used as the time that  turn on a relay.

But, I think that this PID class calculate the amount of variation of output.
So, I feel something’s strange.

orgua, 2016-06-22:
In case of trouble you can take a piece of Paper and try to match the given pseudocode in your given WikiLink and the code shown here: http://brettbeauregard.com/blog/2011/04/improving-the-beginner%E2%80%99s-pid-derivative-kick/
You will see that everything is fine

nanahuse, 2016-06-22:
Thanks. I tried to match as you say. and I understood that this library is not wrong.
But still, I feel strange that the value calculated by PID class is used as output on examples.

br3ttb, 2016-06-22:
so I looked at the discrete link you reference,  and it does not do output = old_output+..

it stores the error to compute the derivative next time, but the output is calculated in a similar fashion to the pid library.

while neither the pid library nor the link you sent do this, there IS a form of the pid algorithm that computes deltaOutput.  it is known as the velocity form.  by taking the derivative of the standard pid equation, you get:

dOutput = kp* dErr+Ki*err+Kd+d2Err

this may be what were thinking of when you asked your question

closed 2016-06-22 by br3ttb

brandoncurtis, 2016-06-24:
@br3ttb, @nanahuse is definitely referring to the **form** of the discretized PID algorithm, while previous issues are concerned with the **approximation technique**.

From K.J. Astrom's "PID Controllers: Theory, Design, and Tuning" (2nd Edition)
Section 4.6, Page 93: Digital Implementation

e.g. regarding the D term: 
>The approximations of the derivative term are stable only when |a_d| < 1. The forward difference approximation requires that T_d > Nh/2. The approximation becomes unstable for small values of T_d. ... The backward difference approximation will give good results for all values of T_d. ... The forward approximation is seldom used because of the problems with instability for small values of derivative time T_d.

Regarding **form**:

>The algorithms described so far are called **positional algorithms** because they give the output of the controller directly.  In digital implementations it is common to also use **velocity algorithms**.  The discrete time version of such an algorithm is also called an **incremental algorithm**. This form is obtained by computing the time differences of the controller output and adding the increments.

>In some cases integration is performed externally. This is natural when a stepper motor is used. The output of the controller should then represent the increments of the control signal, and the motor implements the integrator.  The increments of the proportional part, the integral part, and the derivative part are easily calculated from <equations>.

>One advantage with the incremental algorithm is that most of the computations are done using increments only. Short word-length calculations can often be used.  It is only in the final stage where the increments are added that precision is needed.  Another advantage is that the controller output is driven directly from an integrator. This makes it very easy to deal with windup and mode switches.  A problem with the incremental algorithm is that it cannot be used for controllers with P or PD action only. Therefore, ΔP has to be calculated in the following way when integral action is not used: <equation>

If you're just going to integrate the incremental algorithm on the board and use the result to actuate, shouldn't the positional and incremental algorithms have the same performance for a given evaluation time interval?

br3ttb, 2016-06-30:
@brandoncurtis They should have identical performance for the most part.  The velocity (incremental) form has fewer bugs out of the box (there's no windup problem for example,) but when compared to the existing library (and the code fixes therein) the performance should be nearly identical.  The only difference I can think of is that it will take on extra cycle at startup to get a full D term, since it needs the last two points to calculate the derivative instead of the last one