br3ttb/Arduino-PID-Library #49: ITerm Implementation

AASanchezA, 2016-08-22:
related with Issue #23 , 
ITerm implementation  currently is
` ITerm+= (ki *error);`
which is ok but there is a Bug if you define ki = 0; 
because the implementation of the output
```
  /*Compute PID Output*/
      double output = kp * error +  ITerm - kd * dInput 
```
the integral term is no equal to zero, because it will keep the values untill the time when your change the Ki parameter

my proposal will be, calculate the ITerm like
` ITerm+= (error);`
and the output would look like this
```
  /*Compute PID Output*/
      double output = kp * error +ki*  ITerm - kd * dInput 
```
which look closer to the PID definition

[image]

wait for comments ;)

br3ttb, 2016-08-22:
this is the expected behavior.  if you change ki to 0 while the controller
is running, the intergral will stop contributing to the change in output,
but it will not cause a jump.  if you would like to have the output jump to
a new value, you can make use of the fact that when the controller is
initialized (man->auto) Iterm is set to the current value of *myOutput.

the reason for the existing code can be found here:
http://brettbeauregard.com/blog/2011/04/improving-the-beginner%E2%80%99s-pid-reset-windup/

br3ttb, 2016-08-22:
> beginner%E2%80%99s-pid-reset-windup/

sorry.  wrong link.  here is the explanation:
http://brettbeauregard.com/blog/2011/04/improving-the-beginner%e2%80%99s-pid-tuning-changes/

AASanchezA, 2016-08-22:
Thanks for the explanation,  clear and precise!!
still if you want to tun the PID using the  Ziegler-Nichols method, you have a unexpected output and you have to reset the arduino and initialize with the proper ki and kd equal to zero

br3ttb, 2016-08-22:
for most processes, you'll want to have the ZN output centered around
something other than 0 (e.g. a reflow oven at 150degrees with a
steady-state output of, say, 30%) .  how would you achieve this with the
code you propose?

AASanchezA, 2016-08-23:
Mathematically, the implementation that I propose will respond the same way as the original one.

my question came from the response that I saw from the PID, when I was tuning it using the Ziegler-Nichols method, but with the explanation from the Link that you send me clarify everithings

closed 2016-08-24 by br3ttb