br3ttb/Arduino-PID-Library #57: linear actuators

stephanschulz, 2017-03-31:
first of thanks for this great library.

i am trying to use PID with a linear actuator that uses a Brushless DC motor. 
i based my code of your adaptive example. when reaching the setPoint the linear actuator slowly down nicely. but i also need it to have a slow start up so that the motor does not jump from 0 to max speed.

do you have any tips on how to make that happen?

thanks.

qrqiuren, 2017-04-01:
Except position, you need another PID to control the speed.

stephanschulz, 2017-04-01:
sorry i am not sure i understand you suggestion.

currently i am getting an output value in the range of 0 -255. if my distance is large this value is high - close to 255 and only starts easing out when we get close to the target position.

if you want me to use a 2nd PID for the speed, how to i make those two PID output values work with each other?

agjunyent, 2017-04-04:
There are several ways to achieve this without a second PID ("ramping" output, limit derivative, tracking, etc). I recommend you to create a simple "ramping" output similar to:
if ((output - last_output) > MAX_RAMP) output = MAX_RAMP
It's VERY easy to do, and if your application does not need very high precision, it's perfect

stephanschulz, 2017-04-04:
@agjunyent thanks for the advice.

ultimately i need to reproduce this wave behaviour
http://lozano-hemmer.com/wavefunction.php

i have it working partially were every actuator is following a master oscillator which runs on it's local micro controller.
but i was hoping with PID i might be able to make the actuators "dumber" and have them just react to a position input which comes in over a network connection (OSC). communication works. 
but since new locations do not come in fast enough the actuators need to create in-between motion.
this is were i was hoping PID would help to apply the right speed and error correct.