br3ttb/Arduino-PID-Library #58: Understanding the Parameters Ki,Kp,Kd

TonyWilk, 2017-04-10:
This may be helpful to anyone trying to understand what the parameters do...

The output of the PID in the Compute() method is:
```
      /*Compute PID Output*/
      double output = kp * error + ITerm- kd * dInput;

```
so, if you print out the values of _Kp*error_, _ITerm_ and _-kd * dInput_ you can see how each parameter makes up the output value.

Start by using parameters Kp=1.0, Ki=0.0, Kd=0.0 then try Kp=0.0, Ki=1.0, Kd=0.0 and so on.

It is useful to note that it is output is just the ITerm when it reaches the setpoint. The PID will always reach the setpoint with only a Ki parameter, but it may take some time. The Kp and Kd parameters are used to speed up getting near the setpoint.

One way to simply do this is...
in PID:Compute(), after /* Compute PID Output */ add the following:
```
     // save each term of the output equation
      dispKp= kp * error;
      dispKi= ITerm;
      dispKd= 0-(kd * dInput);

```
then use the PID:GetKp(), PID:GetKi() and PID:Kd() to get those values in your main code

Yours,
  TonyWilk

edgsc, 2017-09-20:
Didn't really understand how to do it. Could someone explain it again?
Thanks!

patrickstanley, 2017-09-22:
Try this for an explanation of PID parameters. http://survivingtheworld.net/ScienceComic3.html

edgsc, 2017-09-22:
Really nice =), but my problem isn't understanding the PID parameters, but how to DISPLAY each term.

I ended modifying:
dispKp= kp ;
dispKi= ki;
dispKd= kd;

And calculating on Excel the P Term, I Term and D term contribuitions for the output.

But what I really whished was to be able to show each term only with Arduino code.

rtek1000, 2022-04-23:
http://brettbeauregard.com/blog/2009/05/graphical-front-end-for-the-arduino-pid-library/