br3ttb/Arduino-PID-Library #61: Relay Example - Output Drop

jnosky, 2017-06-13:
Line #54
 if (millis() - windowStartTime > WindowSize)
Should be:
 if (millis() - windowStartTime >= WindowSize)

Existing code will drop the relay output briefly when the = event sometimes occurs

edgsc, 2017-09-22:
Where's the error  (what file?)? Couldn't locate it. Wasn't it changed to lastTime and corrected?

jnosky, 2017-09-29:
Relay Example? Line #54?  What do you mean you can't find it?

edgsc, 2017-09-29:
You are right! The original line 54 of Relay Example is like the first one you said and your point makes sense. If millis(s) - windowStartTime were = WindowSize, Arduino would do nothing and "skip a cycle".

Thanks for your contribution, I added it to my experiment too =)!