br3ttb/Arduino-PID-Library #63: please use float instead of double in the API

tofrnr, 2017-09-07:
as you stated already by yourself, real 64bit double is overprecise and overtaxing for PID regulation, and actually on AVR your are using 32bit float instead because double is not existing by 64bit.
But on ARM cores there is a doble data type of 64bit which has to be changed then to float, akward enough.
So why not stay with float for all platforms and cores from the start?

per1234, 2017-10-22:
@tofrnr is the king of duplicate issues!

Duplicate of https://github.com/br3ttb/Arduino-PID-Library/issues/39

tofrnr, 2017-10-22:
> @tofrnr is the king of duplicate issues!

per1234, stick this to you where the sun won't shine :P

aguaviva, 2024-05-17:
What is the advantage of using `double`?