br3ttb/Arduino-PID-Library #69: Can't use two instances of PID at once?

xoxota99, 2018-02-01:
I'm trying to use two independent instances of the PID controller (one for X axis, and one for Y axis) to control a ball-on-plate system (similar to https://www.youtube.com/watch?v=K-F_T59ZDPw&t=74s ). I've found that I can get the X-axis working, or the Y axis, but whenever I enable both controllers, only the first one that has it's Compute() method called will actually return an output. The second will always return a zero value for output. I have no idea why this is.

ccoenen, 2018-03-25:
#45 suggests that this _should_ work.

xoxota99, 2018-03-26:
YES! That is the answer! Would love a note in the README about this easy-to-overlook behavior. (lazy eval / conditional shortcutting). Thanks so much!

closed 2018-03-26 by xoxota99