br3ttb/Arduino-PID-Library #72: output goes crazy as error approaches zero.

xoxota99, 2018-03-22:
My PID output goes crazy as my error approaches zero. What am I doing wrong?
Kp=2
Ki=0.1
Kd=0.3

[image]

br3ttb, 2018-03-25:
Looks like too much derivative.  Try kd=0 to confirm.

br3ttb, 2018-03-26:
> Yeah, the signal is a little noisy in that graph. But in general the
> Derivative works, EXCEPT when error is "too close" to zero. Then the whole
> system will suddenly "kick" (which of course pushes the error back up, and
> starts the whole thing over again).

There is more likelihood of quick back and forth near setpoint; when you're
away it's clear which direction you have to go.  So if your input jumps
back and forth once, it creates a step slope which causes the d term to
spike the output.  If the system is pretty quick, that can cause a steep
input slope in the other direction, and on and on.

That's what it looks like to me anyway.  If you set kd to 0 you can
confirm. If I'm right I see two pretty simple options:
1: don't use d. With a system this fast you can probably get the
performance you need with just p and I
2: filter the signal a little to get rid of the spikes: pidInput =
 (rawInput + 9*pid Input)/10

xoxota99, 2018-03-27:
Thanks a ton for the helpful feedback, I'll definitely give this a try and report back!

br3ttb, 2018-03-27:
A third option I didn't mention, which is similar to the first: use LESS
d.  Add a little to get the oscillation-dampening benefit, but not so much
that you get twitchy.