br3ttb/Arduino-PID-Library #73: Example code produces incorrect results.

mike-matera, 2018-04-10:
Hello! 

I'm the developer of FastPID (https://github.com/mike-matera/FastPID). I've developed simulation models to test the correctness of my controller. As a part of that process I've tested its performance against your PID and I've found errors in your example code. 

Here's how ArduinoPID performs against a reference implementation: 

[image]

The problem is that your PID controller produces differential values, but you are not summing those values into an accumulated output. The problem can be fixed in PID_Basic.ino by the following change to loop(): 

```c++
void loop()
{
  static unsigned long int accumulator = 0;
  Input = analogRead(PIN_INPUT);
  myPID.Compute();
  accumulator += Output
  analogWrite(PIN_OUTPUT, accumulator);
}
```
While your PID implementation is correct according to references like Wikipedia, those control systems are designed to actuate a physical valve. The PID output drives a motor that changes the valve and the state of the valve is a form of memory. The ```analogWrite()``` function has no memory so the valve state is simulated with the ```accumulator``` variable. 

It is very likely that many of the tuning and user issues are related to this problem.

br3ttb, 2018-04-10:
two things:
1. where did you get the idea that the library outputs a differential? it
doesn't.
2. to make sure there are no other misunderstandings, can you confirm that
you've read my detailed blog series
<http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/>
where I explain the code? once we're sure we're on the same page, we can
dig down and figure out why the library isn't behaving in the way you
expect.

mike-matera, 2018-04-10:
Hello! Thanks for getting back to me. 

I have read your blog extensively. I used it as a guide when I developed my own implementation. It is amazingly helpful and I really appreciate the time you've spent making the concepts clear. 

I got the idea from reading the code. The Compute() function uses the ```output``` local variable to do intermediate calculations, then assigns the ```myOutput``` pointer. So the output assigned is P + I + D which is the relative output for the timestep (in the diagram u(t)). It matches this model on Wikipedia: 

[image]

I don't see the actuator state anywhere. It's not in the example code. Did I miss something?  

How do you verify your controller? I will be happy to run your verification code and see if I'm doing something wrong. Check out the ./test directory in my repo. My code basically does this with your controller: 

```c++
double Setpoint, Input, Output;
PID pid(&Input, &Output, &Setpoint, 0, 0, 0, DIRECT);

// This is done once   
  pid.SetTunings(kp, ki, kd);
  pid.SetOutputLimits(outmin, outmax);
  pid.SetMode(AUTOMATIC);
  pid.SetSampleTime(1000); 

// This is done repeatedly
  pid.Compute();
``` 

You can see exactly what it does here: 

https://github.com/mike-matera/FastPID/blob/master/test/arduinopid_wrapper.cpp

My verification system is mixed Python/C++ and it emulates the Arduino library. There is exactly one function emulated, ```millis()```, which simply returns 1000 more than the last time you called it. I have a reference PID written in Python that's used to generate the reference signal: 

https://github.com/mike-matera/FastPID/blob/master/test/refpid.py

If you like I can help you get the test code running. You'll probably need a Linux machine or VM.

br3ttb, 2018-04-10:
so in the wikipedia diagram, the 3 terms add up to u(t):  the output at
time t.  if those terms were computing the change that would read du/dt.

there's definitely a popular way of implementing the pid that does that.
the position form (which is more-or-less what I use) is:

U = Kp e(t) + Ki *sum(e) + Kd de/dt

the one the computes the differential is the derivative of the above:

dU/dt = Kp de/dt + Ki*e(t) + Kd d2e/dt2

the "Actuator state" you're looking for is actually the hidden in the
integral term.  look at the position form above.  if error and de/dt are 0,
you can still get a non-zero output if sum(e) is non-0.  and that's what
you'll find.   if you're at steady state (0 error,) sum(e) will be = U/Ki.

br3ttb, 2018-04-10:
...and I keep forgetting to mention,  I have a tester as well:
https://github.com/br3ttb/arduino-pid-library-tester
I run any new updates through it to confirm the library's performance
hasn't changed.

mike-matera, 2018-04-11:
I also use the position form (my PID is based on your work afterall). So if, 

u(t) = Kp e(t) + Ki * sum(e) + Kd de/dt 

And the factory process is trivial, as it is in my test program: 

y(t) = u(t)  

Then a P controller (a controller with Ki=0 and Kd=0 will) never converges on the setpoint. But, it should. Here's what I get when I simulate your controller with only a P term: 

[image]

Stepwise, the following happens (starting with step 3, when the setpoint changes) : 

Kp = 0.5
e(3) = 100, u(3) = 0.5 * 100 = 50  
e(4) = 50, u(4) = 0.5 * 50  = 25
e(5) = 75, u(5) = 0.5 * 75 = 37 
e(6) = 63, u(6) = 0.5 * 63 = 31
e(7) = 69, u(7) = 0.5 * 69 = 34 
e(8) = 68, u(8) = 0.5 * 68 = 34 

Am I missing something?

br3ttb, 2018-04-11:
.Awesome. I think we got to it: A p-only controller WILL NOT converge on
the new setpoint. That's the whole problem, and why you need integral in
the first place: a phenomenon known as *offset*.

With a p-only controller, the equation is:  u=kp*error.

When you start,  error is 0, and output is 0. Great!  But here's the
thing... ANY time the error is 0 the output is 0. And for your system (and
all self-regulating systems) that means a resulting input of 0.  So, you
start with a 0 setpoint and step it to 1.  This gives you some error and
the output jumps.  But as the input starts rising, error (and your output)
start to decrease.  You ultimately reach a steady state where you have just
enough error to lead to an output that gives you that error again. that
error is called offset.

If you decided to treat the output from the above equation as a
differential you could certainly get rid of the offset, but strictly
speaking it would no longer be a pid algorithm.

Now, on to integrating processes (tank level, linear slide position,) where
offset exhibits itself differently.  There, you can totally get away with a
p-only controller. For these, instead of the trivial process being y=u,
it's dy=u; the output controls the rate of input change.

With those guys, so long as 0 output corresponds to no movement, p-only is
adequate (PD is often used as well.) You coast in to your new linear slide
setpoint, error drops, output drops to 0, and you stop moving. Perfect.
The issue arises when you need to handle a load change.  If someone starts
pushing on your slide and you need to resist, you're screwed.  You need
error=0 to lead to a non-0 output, and as we saw above with
self-regulating, it's not going to happen.  So your load will push the
slide off setpoint, error will increase, which will allow for increased
resistance.  Offset again.

mike-matera, 2018-04-12:
Yes! You're right. I had misinterpreted something I'd read and really believed that a P controller would converge. So I added what's essentially an integral term to make it so! 

Thanks for setting me straight. Turns out the bug is mine!

closed 2018-04-12 by mike-matera

br3ttb, 2018-04-12:
No problem Mike,  as soon as you said "yeah of course I've read your blog"
I knew I was in for a fun conversation.  And to give you some credit, a
P-Only controller will converge on some processes (which is probably how
the idea got planted.) so yeah.  I guess the bug was that y=u should have
been dy=u.

In that vein,  take a look at my simulator (mentioned above); it does a
less-trivial model (with deadtime, etc.)  two models actually.  You can
pick between regular or integrating.  might be worth rolling in to your .py

mike-matera, 2018-04-12:
I'll take a look. The great thing about Python is that it's easy! I fixed my reference model and made my controller match. Here's what I get now. 

[image]

My test code has a mode where it applies random parameters and tests against the reference model. It computes the sum of square errors and looks for parameter combinations that are outliers. I'm going to rerun it against my code I'd be happy to report anything I see in yours. 

Thanks again!