br3ttb/Arduino-PID-Library #79: PID library execution and wierd spikes in Signals

KingZuluKing, 2018-05-10:
Hi Brett, since couple of weeks i'm working to understand the reason of the spikes in my PID controlled system but, without any success. As someone highly related to PID, maybe you could put some light into this darkness. Please have a look at the graph I plotted using the data from my Arduino Due. What could be the issue here?, maybe the function which drives the setpoint variable up and down? 

[image]

br3ttb, 2018-05-11:
The pid library doesn't write to the setpoint or input. Only the output. If
your setpoint is moving it's because of something else.

KingZuluKing, 2018-05-11:
I have written such a function block inside my code to act as a setpoint. Actually it just simulates the change in Setpoint variable, in order to see how the system reacts to that. Could you please have a look at it. I cant see what is wrong there, also what might be related to that spikes in setpoint, the graph above.

mike-matera, 2018-05-13:
Here's some feedback on your function. Your loop is being run every 1 or more millisecond. This statement doesn't make sense: 

```
if (current_millis - start_millis >= hold_1)
```

```current_millis``` and ```start_millis``` are both integers. Their difference is either 0 or exceeds ```hold_1``` which is a float with a value 0.01. Along those same lines, this statement has no effect:

```
 start_millis += hold_1;
```

Again, ```start_millis``` is an int and ```hold_1``` is a float that's assigned to 0.01. From your graph it seems like everything is working fine until around 500ms into the experiment. The behavior can't be explained based on the code you've shown. Look for other places where the setpoint is referenced. It's possible that you have a memory error (like overwriting the bounds of an array) that could be affecting setpoint too.

KingZuluKing, 2018-05-16:
[image]
I was just wondering how the desired voltage profile can be programmed, so that it has a square wave geometry in graph the Violet line. It might be helpful to troubleshoot the issue. Any ideas would be of great appreciation.