br3ttb/Arduino-PID-Library #82: Error while compiling the example

set4up, 2018-09-14:
I downloaded and installed the library, and on all examples gives an error:
C:\YandexDisk\arduino-1.0.6\libraries\PID_v1\PID_v1.cpp: In constructor 'PID::PID(double*, double*, double*, double, double, double, int)':
C:\YandexDisk\arduino-1.0.6\libraries\PID_v1\PID_v1.cpp:46: error: type 'PID' is not a direct base of 'PID'

2567366108, 2019-11-30:
same problem

softm39, 2021-12-06:
Same problem 1.0.6 
Was fixed ever?

softm39, 2021-12-06:
This helps, i.e. not declare as code definition (not works seems)
but just run

/*Constructor (...)*********************************************************
 *    To allow backwards compatability for v1.1, or for people that just want
 *    to use Proportional on Error without explicitly saying so
 ***************************************************************************/

PID::PID(double* Input, double* Output, double* Setpoint,
        double Kp, double Ki, double Kd, int ControllerDirection)
{
     PID(Input, Output, Setpoint, Kp, Ki, Kd, P_ON_E, ControllerDirection);

}