br3ttb/Arduino-PID-Library #88: outputSum Intialization causes PD Controllers to have Permanent Offset

aaron9412, 2019-02-16:
Brett,

In PID::Initialize(), outputSum is set to whatever the current output is.
If not using P_on_M or an integral portion of the controller, outputSum will never change.
If output is nonzero when intialized, this will permanently offset the output of the controller.
Also, for a small integral constant ki, this will leave the controller with a large offset to overcome.
I would recommend changing:
>  outputSum = output;

to:
> outputSum = (outMin + outMax) / 2

in PID::Intialize().

Full PID::Initialize() function for reference:
> void PID::Initialize()
> {
>    outputSum = output;
>    lastInput = *myInput;
>    if(outputSum > outMax) outputSum = outMax;
>    else if(outputSum < outMin) outputSum = outMin;
> }

br3ttb, 2019-02-16:
this is intentional. for the vast majority of cases you want "bumpless
transfer". if a car was perfectly holding 60mph when you initialized cruise
control with a 60mph setpoint, what would you want the output to do? Jump
to some arbitrary value? Or stay exactly where it was.  The pid library
does the latter.

The good news is that you can just set the output variable to whatever you
want (presumably 0 in your case) before you call setmode(auto) and the pid
will then work from that baseline.

closed 2019-02-16 by br3ttb

aaron9412, 2019-02-17:
Oh! You're entirely right.
I initialized my outputs at a non-zero outside of the PID limits.
All I need to do is initialize them at 0.

Thank you, and thanks for the library!