br3ttb/Arduino-PID-Library #89: PID_v1 Timers & Interrupts

CarlP66, 2019-03-13:


CarlP66, 2019-03-13:
Hi. Does PID_v1 use timers or interrupts? I'm guessing that it does. The reason for the question is that I seem to have a clash with the NewPing.h library which I believe uses timer 2 (and maybe 4). If I put NewPing calls inside a PID loop, then the PID controller seems to behave erratically. Unfortunately, I can't post the code, as I didn't keep a copy.

br3ttb, 2019-03-13:
I doesn't. it just looks at millis. every time the compute function is
called it checks the time and decides if it should evaluate.

Because of this, it's recommended that compute be called every cycle of
loop(), and that there are no delays in the loop function.

Erratic behavior can occur if this recommendation isn't followed.

CarlP66, 2019-03-13:
Thanks for a quick response. It is an interesting answer. I will have to dig deeper. Cheers.