br3ttb/Arduino-PID-Library #93: Using the library with different input/output

Tommrodrigues, 2019-05-18:
Hi there, I'm looking for some help.

I'm currently developing a script [here](https://github.com/Tommrodrigues/homebridge-nodemcu-boiler/blob/master/NodeMCU-Boiler.ino) which takes the ambient temperature in the room and then adjusts the boiler (central heating) temperature so that it warms/cools the room to the correct temperature.

At the moment, I'm using a PID function I found on a similar project online to do this calculation, and whilst it works most of the time, I get issues occasionally where the function seems to stop reacting and forces me to restart the Arduino in order to get it working again. This is why I'm looking to use this library.

As I understand it, however, this library matches the input to the output. This would not suit my application though as, say for example, the current room temperature is 22 Celsius and I want it to be 24, the boiler would need to be set to a high temperature like 84 Celsius in order to make up this difference then as the ambient temperature nears 24, the boiler would slowly ramp down its temperature in order to not overshoot and then maintain the temperature. 

I would really appreciate some guidance on how exactly I can use your library to achieve this as I'm struggling to wrap my head around how your library could convert the difference into a boiler temperature in this way - how would I get `Output` to be translated to a boiler temperature? Is this even possible?

Thank you in advance for your help,

Kind regards, Tom

br3ttb, 2019-05-18:
this library matches the input to the SETPOINT by adjusting the output.
standard stuff. For you, setpoint is your desired temp, input is actual
temp, and output is your signal to the boiler.

Tommrodrigues, 2019-05-19:
@br3ttb Ah I see, it's that simple! Thanks for your help, I'll look to implement it into the script soon. Is the simple example script included in the repository sufficient to use as a guide for this use-case? Is there a way I can set the bounds for the output?

Tommrodrigues, 2019-05-19:
Got the script working perfectly now. Thanks for your help!

closed 2019-05-19 by Tommrodrigues

br3ttb, 2019-05-19:
should be that easy.

the SetOutputLimits function is what you're looking for. it defaults to
0-255, but it can be any floating-point range

Tommrodrigues, 2019-05-19:
That's great, I've managed to integrate it into [my script](https://github.com/Tommrodrigues/homebridge-nodemcu-boiler/blob/master/NodeMCU-Boiler.ino). Only thing I can't quite work out now is what tuning parameters I should set - I understand it will vary from use-case to use-case and I've had a look [here](https://playground.arduino.cc/Code/PIDLibrarySetTunings/) where you explain what each of the 3 parameters does but I'm still a little lost. For example, if the ambient temperature is not too far off the Setpoint, I don't want the boiler temp to be set to max because it'll likely overshoot. Any idea what sort of parameters I should put in? Is it worth implementing the adaptive parameters as per your example for this case?