br3ttb/Arduino-PID-Library #104: Changing sign Derivative factor

DavyEHB, 2020-04-24:
I compared different versions of the PID library.
I noticed that the sign of the derivative factor changes between the versions.
In the code on your website
http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/
```
/*Compute PID Output*/
   Output = kp * error + ki * errSum **+** kd * dErr;
```
http://brettbeauregard.com/blog/wp-content/uploads/2011/03/pidalgorithm.png
Also includes the addition of the derivative.

v 1.2.1
Here you subtract the derivative.
```
      /*Compute Rest of PID Output*/
      output += outputSum **-** kd * dInput;
```
Quite confusing to me. Since it is negative, won't it counteract the response?
Could you explain the logic for this?

Kind regards.

AnderbergE, 2020-04-28:
I think what you are looking for is explained in the following post:
http://brettbeauregard.com/blog/2011/04/improving-the-beginner%e2%80%99s-pid-derivative-kick/

mrx23dot, 2026-01-30:
please close ticket