br3ttb/Arduino-PID-Library #105: detailed documentation missing

shepner, 2020-09-12:
The detailed documentation page (http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/) is not accessible anymore.  Would you please move it to here?

MonarchActual, 2020-09-19:
I can still access the page you mentioned and there is the documentation on the Arduino Playground (https://playground.arduino.cc/Code/PIDLibrary/).

br3ttb, 2020-09-19:
Which link? The link in the OP works

shepner, 2020-09-19:
Well, never mind.  It wasn't working last week but I can get to it now

br3ttb, 2020-09-19:
yeah it was down for a few hours. Database password issue

simeon9696, 2021-06-29:
It seems like it's down again

rtek1000, 2022-04-23:
I noticed that the theory blog is very old (year 2011) and now I saw this issue about website unavailability. (I already had this kind of problem with an AC power control library, it was a library that was only referenced on the Arduino website, and then the website that hosted it went down and was not raised anymore, the library disappeared to I never found this one anymore.) And I also noticed that the blog with the theory still has "http" instead of "https", and the links on the Aruino website are out of date. I saved these posts using Firefox V99, and I leave it attached here in case the blog goes down and doesn't come back, I think this theory is important to know how to get the best performance out of the library.

[Websites Auto Tune Library Theory.zip](https://github.com/br3ttb/Arduino-PID-Library/files/8546736/Websites.Auto.Tune.Library.Theory.zip)