br3ttb/Arduino-PID-Library #106: P_ON_M functionality doesn't seem to do what it's supposed to - same as D

louis-potvin, 2020-10-04:
Hi,

Looking at Brett's [blog about Proportional on Measurement](http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/), I see that when used, the gain induced by Kp should be `Kp[input(t)-input(init)`. However, looking at the code, it seems it's actually not doing that, but rather having the same effect as Kd:

**Kp Gain:**
`      if(!pOnE) outputSum-= kp * dInput;`

**Kd Gain:**
`      output += outputSum - kd * dInput;`

Really appreciate if someone could tell me what I'm missing, or if there is indeed a discrepancy between the code and the blog, and in that case, which is right :)

Thanks!

br3ttb, 2020-10-04:
take a look at the next post:
http://brettbeauregard.com/blog/2017/06/proportional-on-measurement-the-code/

louis-potvin, 2020-10-04:
Thanks for the quick reply Brett. I realize I fundamentally missed the part where the gain of Kp goes in outputSum and therefore is "memorized" between iterations, whereas the gain of Kd does not. 

And thanks for the library!

closed 2020-10-04 by louis-potvin