br3ttb/Arduino-PID-Library #112: how to reset PID after runing

gunaone, 2021-06-11:
when i run it, at Output 255 and i stop execute it, how i can reset it to 0 again. i try with still running but set input high, its not enough fast to get down to 0 again.

much thans,

br3ttb, 2021-06-11:
when the pid turns on, it initializes based on the current value of the
output variable. So this should work:
Pid.setmode(manual)
Output=0
Pid.setmode(automatic)

gunaone, 2021-06-13:
thanks @br3ttb for your replay

so ,
i just execute this code once :
`void reset_PID()
{
  Pid.setmode(manual);
  Output=0;
  Pid.setmode(automatic);
}`

br3ttb, 2021-06-13:
assuming that "Output" is the name of the variable you linked with the pid
in the constructor, and you want to drop it to 0, then yes.

If you have any error (setpoint!=input) then the output might not be
exactly 0 at startup, because of the pterm contribution.

gunaone, 2021-06-17:
thanks @br3ttb ,
it's very helpful...
😊😊😊✔

closed 2021-06-17 by gunaone