br3ttb/Arduino-PID-Library #117: PonM vs PonE coeffitients

djObsidian, 2022-01-28:
Are PID coeffitients for PonM and PonE equal? Or they differ? 
I'm sure that all pid tuning methods do work for both, but i'm trying to synthesize coeffs. instead of picking them up empirically.
Will this https://www.mathworks.com/help/control/ref/pidtuner-app.html produce valid PID coeffs. for PID with PonM?
If not, how should i calculate coeffs. from system step response?

rtek1000, 2022-04-23:
https://github.com/br3ttb/Arduino-PID-Library/issues/77

rtek1000, 2022-04-23:
http://brettbeauregard.com/blog/2012/01/arduino-pid-autotune-library/

rtek1000, 2022-04-23:
http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/

rtek1000, 2022-04-23:
http://brettbeauregard.com/blog/2017/06/proportional-on-measurement-the-code/

djObsidian, 2022-04-23:
Thanks, i read all of this. However, knowing how PonM works doesn't helps tuning it. The most common method, Ziegler–Nichols, simply doesn't works here, because no oscillations will occur with Kd=0, Ki=0 and Kp set to anything with PonM. PID autotune also doesn't works with systems with large time constant, like several minutes.

rtek1000, 2022-04-23:
From what I understand, if the instructions for use don't talk about the input declaration of these coefficients, it's because it's enough to activate one mode and deactivate the other.

> How to use it in the new PID Library
> 
> If you’re ready to try Proportional on Measurement, and you’ve installed the latest version of the PID library, setting it up is pretty easy. The primary way to use PonM is to specify it in the overloaded constructor:
> 
> [image]
> 
> If you’d like to switch between PonM and PonE at runtime, the SetTunings function is also overloaded:
> 
> ![SetTunings
> ](http://brettbeauregard.com/blog/wp-content/uploads/2017/06/SetTunings.png)
> 
> You only need to call the overloaded method when you want to switch. Otherwise you can use the regular SetTunings function and it will remember your choice.

Source: http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/

- If leaving ki = 0 and kd = 0, the system cannot oscillate:
- - Did you try to increase the kp value?

- If Autotune can operate with PonE:
- - Would it be possible to tune with PonE on and PonM off, so save these parameters?
- - And use the same parameters with PonM instead of PonE?

I found this explanation:
(https://www.google.com/search?q=how+to+tune+pid+large+dead+time)

> So hard and the simple way
>
> In a dead time process, the controller makes a change, then waits, waits, waits until the dead time has passed. Finally the controller finds out how its change effected the process variable. It is like trying to drive a car blindfolded with the passenger telling you what to do. You will need to go very slowly. The first try at getting better control in a dead time process is to reduce the dead time. Simply moving a probe closer to the valve may do it. But sometimes there is nothing you can do to lower dead time. So what can you do?
>
> The advanced controller that is proven to work very well on dead time processes is a Proportional Integral controller with the combination of special stealth tuning parameters. It is important to not use derivative unless there is some lag or first order response in the process. Using derivative on a dead time process would make the control loop unstable.

> Stealth tuning
> 
> This stealth tuning applies to dead time processes. If the process has a lag, then this tuning still applies if the lag is 1/5 or less than the size of the dead time. The stealth tuning uses the process dead time and the process gain:
> 
> controller gain = 0.3/(process gain)
> integral time = .42 * (process dead time)
> 
> The integral units are in time - either minutes or seconds. These equations give optimal tuning -- minimum error to load upsets. If you want slower response, with more robustness, simply lower the gain some. (Note that these equations apply for usual ideal and series PI algorithms. For the parallel algorithm you've got to divide or multiply the integral time by the controller gain.) Controller algorithms are the subject of another presentation ["Differences in PID Algorithms and Units"](http://www.expertune.com/r2.asp?f=autoTechies&d=Jul04&l=present.asp?name=pidalgo) available from the articles page on ExperTune.com.

Source: https://www.automation.com/en-us/articles/2003-1/how-to-control-a-process-with-long-dead-time

rtek1000, 2022-04-23:
This other website seems to complement some details:
- Measuring the Dead Time and Time Constant of a Process
[image]
- Dead Time versus Time Constant
[image]
(This image needs a light colored background to see the details)
- Controller Modes
- Applicability of Tuning Rules
- Controllability
- Loop Settling Time

> Dead Time versus Time Constant
> 
> The dynamic response of self-regulating processes can be described reasonably accurately with a simple model consisting of process gain, dead time and lag (time constant). The process gain describes how much the process will respond to a change in controller output, while the dead time and time constant describes how quickly the process will respond.
> 
> Although the dead time and time constant both seem to describe the same thing, there are several fundamental differences between how dead time and time constant affects a control loop. The first difference is that dead time describes how long it takes before a process begins to respond to a change in controller output, and the time constant describes how fast the process responds once it has begun moving.

Source: https://blog.opticontrols.com/archives/499

drf5n, 2023-03-11:
> * If leaving ki = 0 and kd = 0, the system cannot oscillate:
> * * Did you try to increase the kp value?

Increasing kp in P_ON_M mode with ki=0 and kd=0 won't oscillate because 1) the proportional term acts on the integral in a retarding/negative sense and 2) Assuming that it reaches setpoint, the amount that P_ON_M Kp contributes to the integral/outputSum is the constant -kp*(Setpoint - InitialPV), and with ki=0, there's nothing to move the outputSum away from there. 

With P-only P_ON_M, assuming your process ultimately needs an Output of 42 to maintain a PV at 100 and and you are starting from 25,   With P_ON_M mode with Ki=Kd=0, you need to set kP = -.56 because your initial error you need to integrate away is -75, the ultimate output is 50, so kP = -42/75, which is negative, so REVERSE.   If you tried DIRECT with any positive kP, it would push Output lower than initial.  If you tried REVERSE with lower than 0.56, the process would stop changing (and integrating Kp) before it got to SP.  With higher than Kp=0.56, it would keep moving until the measurement was whatever an Output of kP*75 would produce.  P-only P_ON_M won't work for tuning.

For more tuning more difficult systems, one would make a reasonable/smalll manual step adjustment of the output and see the response, rather than let an automatic control mechanism moderate the response.

br3ttb, 2023-03-12:
If you don't have Ki with a p_on_m controller, you will have an algorithm
that doesn't care about your setpoint at all.  Error=sp-input.  If p and d
are only looking at the input (as is the case with p_on_m,) the i term is
the only one saying "hey guys, we're X away from target. I propose we move
the output by Y"

djObsidian, 2024-01-05:
That's all understandable, the question was something else entirely. Can I tune a classical PID (PonE) using any of the existing methods and then put the same coefficients into a PID with PonM and it will work just as efficiently? Or will it not and the coefficients need to be changed? If the optimal coefficients for PonE and PonM are different, is there a dependency between them?

ariefadha, 2025-08-13:
> > I try to plot the PonM, and the [blog](http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/) has wrong graph on it.
> 
> I understand your confusion! If you try to plot the i term the same thing will happen. Ki * error=0 at the end too! So what gives? The key is the -= for kp and += for ki. We're adding to (or subtracting from) wherever was calculated last time. If you want to duplicate my plot, you'll need a new variable called "displayPTerm" or something. If you do "displayPTerm-=kp * dInput;", the graph of that over time should match the blog post

1st pardon if im being rude. I have limited in English.

just for discussion have you try to read the [discussion ](https://github.com/CapnBry/HeaterMeter/issues/42)that I link above?

and if I plot I-term Ki * error = 0 at the end its true, thats exactly what I mean. and  I-Term need to 0 if output=setpoint.

so the plot with PonM :
Kp * lastinput-inpunt = 0 and  Ki * error = 0. so the output will be 0 

[image]

lets focus on pink box. on input/setpoint diagram when last input = input, then dInput = 0. so in diagram P-term should back to zero (as I draw with thin line purple)

and it will worst if i set integral very low or 0. 

so if "displayPTerm-=kp * dInput;" ...  whereas "kp*dInput = 0" ... so "displayPTerm = displayPTerm-0"

maybe you can enlight me more about this. I also do manual plot in paper for 4 hours just because is really promising equation.
I recently implement my PID equation to force I-Term only kick when input=lastInput. so i can have snappy output with aggresive I-Term and never overshot. but thats is another topics