br3ttb/Arduino-PID-Library #122: Is the calculated PID result relative to the previous value or absolute on its own?

happymacer, 2022-05-02:
Hi Brett, first thanks for your trouble to do this.  Im writing code to make a treadmill motor run at constant speed irrespective of the load.  The code is for a PIC16F690 and my work so far is detailed on my github https://github.com/happymacer/PF906-treadmill-motor-controller- (Ive yet to upload the first version of code )

I can run the motor without PID at the moment by setting the duty cycle and the value of the duty cycle is an absolute value rather than an increment from the prior speed setting (Im using fixed speed increments selected by button presses).  The PID is supposed to keep the motor at the set speed as the motor load increases.

I have been implementing your code in C and I have come unstuck because all the PID example code I have seen so far seem to be an incremental value, so when I make the output of the PID the duty cycle the first time will be fine due to the large error, but the next will be a small error and hence a small duty cycle so when i plug that into the registers for the PWM, the motor will slow right down.

So Im thinking I need to add the new PID output to the previous and drop that into the PWM registers.  Is that correct or have I messed up?

rjjrbatarao, 2022-09-30:
I saw this on the library myPID.SetControllerDirection(DIRECT);  this is incremental if you need the inverse change DIRECT to REVERSE

drf5n, 2023-03-11:
If you need relative output, look into the Proportional On Measurement setting:

http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/