br3ttb/Arduino-PID-Library #123: Is it possible to use the PID inside another class?

studi1882, 2022-05-11:
Hi everyone,
I have to do some pre- and post calculations which I want to do in a class. Within this class I want to use the PID class. If I use the PID package inside a sketch everything works fine.
But when I use it inside a class it does shows following error:
"no match for call to '(PID) (double*, double*, double*, double, double, double, const PID::Direction&)'"

My files look approximately:

temp_controll.h:
#include <PID_v2.h>
class Temp_controll
{
public:
Temp_controll(double now_temperature, double set_current, double wanted_temperature)
	{		my_PID(&my_now_temperature, &my_int_current, &my_wanted_temperature,(double) 2,(double) 5,(double) 1, DIRECT)}
PID my_PID;
};
Because out of the 7 variables the first 6 seem to fit, I was wondering if it might be the last variable (the direction) which is wrong, because it is referenced. I didnt figure out a way to fix it.

I would be really happy if you could help me and thanks in advance!

Vheers

Paul

P.S.: I hope this is the right place to ask such a question if not, I am happy if you tell me the right place.

TuxLeon, 2022-05-12:
The problem is the way you instantiate your PID.

The example should fix your compiler error.

class Temp_controll
{
    public:
    Temp_controll(double now_temperature, double set_current, double wanted_temperature)
    { 
        my_PID = new PID(&now_temperature, &set_current, &wanted_temperature,(double) 2,(double) 5,(double) 1,DIRECT);
    }
    PID *my_PID;
};

studi1882, 2022-05-13:
Thank you a lot for your fast and helpful answer. 
That solved my problem! For the next classes I know how to do it!
Cheers and thanks again :)

closed 2022-05-13 by studi1882