br3ttb/Arduino-PID-Library #124: Output goes to 0 when above setpoint.

ZachElkins, 2022-09-12:
I am running in to an issue where the output is always 0 when the input is greater than the set-point, but seems to be working fine when it is below it.

I think at least one other person is having the same issue, found in the Google Group. (https://groups.google.com/g/diy-pid-control/c/Z9zokl6ngZQ)

heaterC, 2022-10-02:
As you can find in the documentation, the output window is by default 0-255, but you can easily change this.

drf5n, 2023-02-22:
Yes, the default output range is set to the analongWrite(...) range of 0-255.  If you want it to go negative, use SetOutputLimits:

https://playground.arduino.cc/Code/PIDLibrarySetOutputLimits/