br3ttb/Arduino-PID-Library #128: need esp32 PID example

microElabDevelop, 2022-12-26:
need PID example on esp32 controller, i want to drive motor with pot. postion controller(soler tracker)

JacobChrist, 2022-12-26:
https://github.com/br3ttb/Arduino-PID-Library/blob/master/examples/PID_Basic/PID_Basic.ino

microElabDevelop, 2022-12-28:
@JacobChrist thank for responce , i try this but this example not working on ESP32 controller.  i need esp32 example.

thricebeau, 2022-12-28:
@microElabDevelop have you configured your pot pins in the setup and have them update values in the loop to assign to the Input variable? Any code?

microElabDevelop, 2022-12-28:
Now i run just basic pid code . After that use for soler tracker . First i just need simple pid code on esp32 . I try but i think this lib is not for esp32 thats why i searching esp32 compatible lib ..

FelixWT, 2023-02-15:
I don't think it's library's problem but if you really want to try something else check this out
https://github.com/Dlloydev/sTune
It works on my esp32