br3ttb/Arduino-PID-Library #129: basic example uses analogWrite(outputpin,DOUBLE) !

StefanL38, 2023-01-04:
Hi everybody,

the example
https://github.com/br3ttb/Arduino-PID-Library/blob/master/examples/PID_Basic/PID_Basic.ino
defines output as double (=double-precision **floating**-point)

```
//Define Variables we'll be connecting to
double Setpoint, Input, Output;
```

and then uses this floating-point-variable in analogwrite

```
analogWrite(PIN_OUTPUT, Output);
```

that is a BIG BUG !

JacobChrist, 2023-01-04:
I'm curious as to what problem this causing?  Most of the time analogWrite is not better than 12 bits and a single precision float has 23 bits of mantissa.

ottobonn, 2023-01-05:
When you pass a floating point variable as a function argument where the function expects an integer, the compiler will automatically promote the float to an integer as if you had assigned a float to an integer variable. You can find details here, section 6.5.2.2 paragraph 7: https://stackoverflow.com/questions/1255775/default-argument-promotions-in-c-function-calls

When you assign a float to an integer variable in C, the result is that the fractional part of the float is discarded (truncated), so the number is rounded down. If that truncated result is too large to be represented by the integer it's being assigned to, the result is undefined behavior, which generally means it depends on the compiler. https://learn.microsoft.com/en-us/cpp/c-language/conversions-from-floating-point-types?view=msvc-170

In this sketch, if we assume the values of the float don't get larger than the integer type can represent, the result of the type conversion is just that we discard the fractional part of the value when writing it to the pin.

JacobChrist, 2023-01-05:
Another way to put what ottobonn is saying is that the compiler automatically type casts the float to an int when.  So the effective code is:

``` C++
    analogWrite(PIN_OUTPUT, (int)Output);
```

JacobChrist, 2023-01-06:
@StefanL38  My original question to you was "I'm curious as to what problem this causing?".  Because a float or a double can be much larger or smaller than what an int can hold, so when it is converted to an int there could be an issue.  But I suspect that you haven't tried to code because I know it worked for me the last time I tried it.

I do understand why your confused.  But to get the bits to align the and they way your thinking you would actually need to do some funny casting, maybe something like this:

``` c++
int i = *((int *)&Output);
```
The above code is not tested, but this would probably line up all the float bit such that they would not make sense as an integer.  But this is good to think about since if you want to transmit a float over a serial line you can convert it to an integer, then back.

``` c++
float f = *((float *)&i); 
```