br3ttb/Arduino-PID-Library #130: Is there an Overflow in PID_RelayOutput.ino?

rtek1000, 2023-01-12:
Hello,

I noticed that the sum of the value in the windowStartTime variable occurs, but it seems to me that it is not restarted, so in a moment it seems to overflow.

```C++
  if (millis() - windowStartTime > WindowSize)
  { //time to shift the Relay Window
    windowStartTime += WindowSize;
  }
```

Based on the Blink code, the variable can be set using the value of millis(), this way it must always have a value within the operating range:

```C++
  if (millis() - windowStartTime > WindowSize)
  { //time to shift the Relay Window
    windowStartTime = millis();
  }
```

Ref.:

```C++
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
    // save the last time you blinked the LED
    previousMillis = currentMillis;
   [...]
  }
```

https://docs.arduino.cc/built-in-examples/digital/BlinkWithoutDelay

rtek1000, 2023-01-12:
Another thing I noticed in this example:

- WindowSize value range is **incompatible** with Output value range

[PID_RelayOutput.ino (line29):](https://github.com/br3ttb/Arduino-PID-Library/blob/master/examples/PID_RelayOutput/PID_RelayOutput.ino)
`int WindowSize = 5000;`

[PID_v1.cpp (line 28):](https://github.com/br3ttb/Arduino-PID-Library/blob/master/PID_v1.cpp)
```C++
PID::SetOutputLimits(0, 255);				//default output limit corresponds to
												//the arduino pwm limits
```

drf5n, 2023-03-02:
windowStartTime is the same size as the millis() result, so it will roll over at about the same time as millis() itself rolls over

https://github.com/br3ttb/Arduino-PID-Library/blob/9b4ca0e5b6d7bab9c6ac023e249d6af2446d99bb/examples/PID_RelayOutput/PID_RelayOutput.ino#L30

The math for differencing unsigned longs works well for intervals.

Using the `windowStartTime += windowSize;` form makes it so any jitter in the window getting triggered is absorbed by the following window.  The triggers will stay synchronized with every 5000ms for as long as the code runs.   If you use the `windowStartTime = millis();` form, if a window starts a little late, it pushes all the following windows later too.