br3ttb/Arduino-PID-Library #140: esp32 adaptive tuning

tomislav12, 2023-10-19:
double Setpoint, Input, Output;
PID myPID(&Input, &Output, &Setpoint, 0, 0, 0, DIRECT);

setup() {
Input = 10;
  Setpoint = 10;
double Kp = 10; // CHANGED
  double Ki = 0;
  double Kd = 0;
myPID.SetOutputLimits(-200, 200);
myPID.SetSampleTime(16);
myPID.SetMode(AUTOMATIC);
myPID.SetTunings(Kp, Ki, Kd); 
}
loop() {
// below is simplified, in real code it is in some function call):
Input = 5; // (simplified change)
myPID.Compute();
SerialBT.println(Output); // only takes initial params into consideration (0,0,0) ! Why is that?
}

Second question, I need to have params 20,0,-10, can Kd be below zero?

tomislav12, 2023-10-28:
Another question, I found this inside the library code:
/* SetOutputLimits(...)****************************************************
 *     This function will be used far more often than SetInputLimits.  while
 *  the input to the controller will generally be in the 0-1023 range (which is
 *  the default already,) ....
Is input really needed to be 0-1023 or can it be eg 0-12?
Anyway, I cannot find the function SetInputLimits or the default(1023) mentioned in the code of this library...

drf5n, 2024-07-31:
There are no bounds on the input limits. The input need only be in the same units/range to be compatible with Setpoint, so that these calculations make sense:

https://github.com/br3ttb/Arduino-PID-Library/blob/524a4268fc01e6ea397e7fc5b5d820741e9b662f/PID_v1.cpp#L65-L69

This reference to a non-existent SetInputLimits is spurious: 

https://github.com/br3ttb/Arduino-PID-Library/blob/524a4268fc01e6ea397e7fc5b5d820741e9b662f/PID_v1.cpp#L146-L153