br3ttb/Arduino-PID-Library #141: PID Output Relay exemple

sbarabe, 2023-11-01:
Hi,

After experimenting, it seems to me that the relay output in this example is reversed : should be LOW when stated HIGH, and HIGH when stated LOW.

I might be wrong, but seems wrong to me as it is...

Regards and thank for the library !

JacobChrist, 2023-11-01:
HIGH and LOW are dependent on your hardware. A HIGH output could turn a relay on or off depending of how its wired.  The same goes for a low as well.

For example a HIGH driving the gate of an Enhancement Mode N-Channel FET (source = gnd) will allow current to flow (to a relay) and thus turn it on. However a HIGH driving the gate of an Enhancement Mode P-Channel FET (source = Vdd) will turn the FET off and no current will flow (to a relay).  If you need it inverted just put a ! in front of the result.

sbarabe, 2023-11-01:
You're right about this, HIGH or LOW is irrevelant depending on the devices wiring, I see it now. But as an example,  it's easier to understand when HIGH is ON and LOW is off...
But maybe I'm too green to see that easily ! Thanks for the reply !

closed 2023-11-01 by sbarabe