br3ttb/Arduino-PID-Library #143: Link to documentation just takes me to the Arduino Playground homepage

sholtob, 2024-02-18:
The link to the documentation is broken it just leads to the Arduino Playground homepage.

clytle374, 2024-04-17:
Same here, and I desperately need to RT*M

SimonEnsemble, 2024-05-29:
docs are now [here](https://playground.arduino.cc/Code/PIDLibrary/). (thanks, Brett, for pointing this out.)

SimonEnsemble, 2024-05-29:
ha ha---the website URL doesn't save the path... odd.
we need to click "Libraries", then the "PIDLibrary - Provides basic feedback control" is listed on the webpage.

[image]

drf5n, 2024-08-23:
Try with a trailing slash:

  https://playground.arduino.cc/Code/PIDLibrary/

And the explanatory docs at 

* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-introduction/ (Note that the links to the chapters are broken on that page)
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-sample-time/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-derivative-kick/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-tuning-changes/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-reset-windup/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-onoff/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-initialization/
* http://brettbeauregard.com/blog/2011/04/improving-the-beginners-pid-direction/
* http://brettbeauregard.com/blog/2011/04/how-fast-is-the-arduino-pid-library/
* http://brettbeauregard.com/blog/2012/07/beginners-pid-series-translated-to-spanish/
* http://brettbeauregard.com/blog/2017/06/introducing-proportional-on-measurement/
* http://brettbeauregard.com/blog/2017/06/proportional-on-measurement-the-code/