br3ttb/Arduino-PID-Library #147: How can I compute new pid coeff from old ones based on

JulienDevillers, 2024-11-10:
Hello

I'm driving the pwm of an electric motor with a pid controler. 
It works pretty well.

After some testings, I realized that the fact that the motor is powered by a battery
makes the system behave differently depending on the battery voltage.

I will change the system in order to make the pid system control the voltage provided to the motor instead of the pwm ; another
system will do the conversion V -> PWM based on the battery voltage.
Therefore, I expect the same behaviour whatever the battery voltage (if of course, the pid does not ask for a voltage
more than battery voltage)

Concretely, PID will not control values from 0 to 20k (pwm period) but 0 to 8400 (mV).

My question is: 
Is there a mathematical way to compute the new PID coefficients from the old ones to get the same behaviour with the 
new actuator without having to go again to a new pid tuning session?

Thanks.

br3ttb, 2024-11-17:
So an output range of of 0-20k is now 0-8.4k, and the rest of the loop is
equivalent?  If so, then the controller is now 2.4x (20/8.4) more powerful.
That is, an output change of 1 means 2.4x more process change than it used
to.  As a result, dividing the 3 coefficients by 2.4 should make the pid
output with the same oomph as before the conversion

JulienDevillers, 2024-11-19:
Hi Brett,

Thanks for your answer, it was obvious for p, but a confirmation for i and
d was useful!